firmata/protocol #29: Question about midi message format

chellem, 2015-04-17:
Out of curiosity, I am wondering why midi message format has been chosen for communication.  What are the advantages of using it and is there any limitations ?

Regards.

soundanalogous, 2015-04-18:
The original creator of Firmata is no longer involved with the project so I don't know the exact answer but I suspect it is because midi is fairly simple to parse and it requires sending less data than an ASCII-based protocol would. It also makes it easy to frame data. The other common serial protocol I see is to send data with a start byte and ending checksum. This has the advantage of less bit shifting than midi since you send 8-bits at a time rather than midi's 7-bit data. However computing the checksum for each packet I suspect takes roughly as many clock cycles as shifting reformatting the bits in the midi protocol.

damellis, 2015-04-18:
Also, as sensing for music generation (e.g. w/ Max or PD) was one of the original use cases, I think there were concerns about increasingly through-put and reducing jitter by using a binary protocol.

closed 2016-03-06 by soundanalogous