firmata/protocol #31: Analog input functions (amplitude, average, derivative, etc.)

damellis, 2015-05-01:
There are some filters / functions that operate on multiple samples of an analog input and that aren't really feasible to run on the host / computer side. For example, taking the amplitude of the audio signal coming in from a microphone, or the average of multiple samples of a noisy input, or taking the derivative of a quickly changing signal. It would be useful to have a way to process these functions on the microcontroller side.

I'm imagining that the host would specify a function (from a fixed set) and either a number of samples or a total time interval over which to take the samples. Then, I'm imagining that the data would be sent to the computer repeatedly, on the same interval used to report analog input values, until the function was turned off by the host computer. 

What do you think? Is it worth trying to flesh this out in a proper protocol proposal?

soundanalogous, 2015-05-02:
Interesting proposal. This is something I've never thought of adding before. I can see how it could be useful though. I'm a bit concerned about how well this would perform on something like an Arduino Uno that is also trying to communicate with a couple of i2c devices, and a pair of servos all while sampling analog input at a relatively high frequency. Maybe the use cases for filtering don't involved motors and digital sensors though and filtering could be used on it's own with ConfigurableFirmata.

I don't have a good sense for the particulars of a proposal for this feature as I don't use filters much. I'm all for adding this to the proposal. Do you want to take an initial pass?

damellis, 2015-05-04:
There are definitely performance implications, but I think it would be up to the user to specify a sampling interval or quantity that didn't interfere too much with the other things they want to do. My main concern would be finding an implementation that avoids jitter (i.e. different delay lengths on different passes). 

I don't use filters much either but I will try to put together a proposal at some point (unless someone else gets to it first!). My initial use case is determining volume using a microphone so I'll probably start with something that works for that but that could be expanded for other uses.

bmd1103, 2016-11-10:
A related issue is when using a sensor such as an IMU where processing is carried out in the Arduino. eg http://www.freetronics.com.au/products/9-dof-imu-accelerometer-gyroscope-magnetometer#.WCPlQvl97IU.  This generates 9 output floats.