firmata/protocol #37: Documentation of protocol versions > 2.4?

finson, 2015-10-27:
Are there protocol.md files for Firmata Protocol versions greater than 2.4?

soundanalogous, 2015-10-27:
v2.4 is the current version

finson, 2015-10-27:
Right, that's my question. I was hoping to find a protocol.md document that describes all features that are settled parts of the protocol as of a particular version so that if I write a client library I can specify that it conforms to a particular version as documented by protocol-2.x.md.  The current protocol.md document says it defines v 2.4, but it includes the v 2.5 single digital pin feature added in April.

soundanalogous, 2015-10-27:
I need to document this, but any markdown file not prepended by "-proposal" is current. You are correct though in that determining the version is currently confusing. I should have been using tags and will going froward from 2.5. Also I added single digital pin prematurely. Should have kept it beta I added serial.

closed 2015-10-28 by soundanalogous