firmata/protocol #39: Firmata versioning is a mess

soundanalogous, 2015-10-28:
Because Firmata and ConfigurableFirmata diverged a few years ago, they have different version numbers. However as far as the protocol is concerned they should be equal (in that both versions of Firmata are based on the same protocol - it just so happens that ConfigurableFirmata supports more features). I'm not sure how to fix this mess other than maintaining separate versioning for the protocol, the StandardFirmata implementation and the ConfigurableFirmata implementation.

The issue is that features may be added to the protocol and an implementation in ConfigurableFirmata but not implemented in StandardFirmata, this the need to bump the version in the protocol and ConfigurableFirmata but not StandardFirmata.

Open to ideas here if anyone has any.

reconbot, 2015-10-28:
This actually sounds like a really good way forward. Unless you're breaking midi spec, you're just going to be sending unsupported messages which will be dropped in older versions of the protocol, and by devices that don't support them. Having 3 version numbers, 1 for each firmware and 1 for how they talk. Seems reasonable to me as long as we're not breaking backwards compatibility.

I'll be honest I had no idea there wasn't 3 different version numbers.

soundanalogous, 2015-10-28:
Digging thorough past commits of the old "configurable" branch in firmata/arduino, it may make sense to update the protocol version as follows:

v2.4: added `Stepper`, `OneWire` and `Scheduler`
v2.5: changed enable digital port reporting and analog port reporting to return value when enabled
v2.6: added `Encoder`
v2.7: added `Serial` and `Set Digital Pin Value`

Then I need a way to report the protocol version vs the software version (StandardFirmata, ConfigurableFirmata, SomeOtherFirmata, etc).

soundanalogous, 2015-10-28:
Then use `protocol version` (0xF9) and the sysex sub-command `firmware name/version` (0X79) properly. Unfortunately this has not been maintained for some years. Hopefully making the switch will not break existing client implementations.

`0xF9` will return the protocol version
sysex `0x79` will return the firmware name and firmware version (not the same as the protocol version going forward).

soundanalogous, 2016-11-14:
Final resolution here:
- The core protocol covers the features described in protocol.md and a single version covers the core protocol (current version is 2.5.1)
- Each optional firmata feature has it's own markdown file and version. Beginning in firmata protocol version 2.6.0 which will add the ability to query supported features and their versions, individual optional features will be versioned. Existing optional feature version numbers will start at 1.0.0.
- The firmata firmware has it's own version and will vary by library (Firmata vs ConfigurableFirmata, etc).

soundanalogous, 2016-11-14:
Next step is to document the comment above in protocol.md