firmata/protocol #55: Servo mode sent values

OlivieriIan, 2016-02-20:
Greeting. I detected a problem when trying to sweep all posible angles in servo mode. The problem is that, for extended analog, the value 128 is wrong. 
I'll attach a file with the values received in the client when gently moving the slider to obtain all the values in the range (and i did it several times to verify that it was no coincidence). I used Device Monitoring Studio to sniff the port.
[servo message analysis.txt](https://github.com/firmata/protocol/files/139102/servo.message.analysis.txt)

So, i used pin 16 (0x10) to test the mode and when i went from 127 to 128 i got:
F0 6F 10 7F F7 (message for 127)
F0 6F 10 **00** F7 (message for 128) <-- this should be F0 6F 10 **00 01** F7

Snap solves this problem by sending a fixed lenght message for each value:
F0 6F 10 7F 00 F7 (message for 127)
F0 6F 10 00 01 F7 (message for 128)

I hope this helps, and any query is welcome.

soundanalogous, 2016-02-20:
This has nothing to do with the protocol. I need to know a lot more about the code you are running in order to help you.

What code are you running on your board?
What Firmata client library (if any) are you using?
What is your application code?

Also this issue should be moved to the appropriate location. If you suspect an error in the firmware, reopen (with appropriate details) either here (if you're using the Firmata library): https://github.com/firmata/arduino

Or here (if you're using the ConfigurableFirmata library): https://github.com/firmata/ConfigurableFirmata

OlivieriIan, 2016-02-20:
You are right, this is not a protocol problem, it's a Firmata_Test problem. I didn't even specify that, i'm really sorry.
I'll close this and move it to correct repository

closed 2016-02-20 by OlivieriIan