firmata/protocol #60: SYSEX commands for reading and writing key-value pairs to EEPROM

monteslu, 2016-05-05:
based on this discussion:  https://github.com/firmata/arduino/issues/257#issuecomment-216350738

Would it make sense to use the otherwise unused (in most firmata devices) EEPROM as a simple key-value store?

soundanalogous, 2016-05-05:
That is one option and it could use EEPROM or other persistent storage (I think some newer boards don't use EEPROM but there is a way to emulate it somehow) - in that the interface would just be for key/value pairs, another is a more generic wrapper for EEPROM. There has also been talk over the years about using EEPROM internally in Firmata to maintain and restore state. Early on (Firmata v1) used EEPROM in this way if I remember correctly. It was removed in later versions for some reason.

soundanalogous, 2016-05-05:
https://github.com/firmata/arduino/issues/237