firmata/protocol #61: Half Duplex Serial

dtex, 2016-06-06:
My use case it to control a Dynamixel servo which uses Half Duplex Serial. The solutions I've found online use a tri-state octal buffer and a single digital pin to control data direction. I can add an expander in Johnny-Five that adds the direction pin and wraps existing serial but I want to make sure that direction pin doesn't belong in firmata.

Thoughts?

dtex, 2016-06-06:
Found a related issue:

https://github.com/firmata/arduino/issues/159

soundanalogous, 2016-06-06:
It depends on how many changes are necessary. If it's simply adding support for half-duplex serial is simply a matter of using the existing Serial protocol and toggling a specified digital pin on each state change then that may make sense to add as an addendum to Firmata Serial protocol definition. In that case any Firmata host that already supports Serial would just need to add an additional parameter to specify the direction pin. However if there are nuances per specific hardware, then it would be better to handle the direction pin separate from the Serial protocol.

dtex, 2016-06-06:
The only thing I've seen in addition to changing the state of the pin is a brief pause (400us) when switching from transmit to receive. 

The Serial Config command would change to:

````
0  START_SYSEX      (0xF0)
1  SERIAL_DATA      (0x60)  // command byte
2  SERIAL_CONFIG    (0x10)  // OR with port (0x11 = SERIAL_CONFIG | HW_SERIAL1)
3  baud             (bits 0 - 6)
4  baud             (bits 7 - 13)
5  baud             (bits 14 - 20) // need to send 3 bytes for baud even if value is < 14 bits
6  rxPin            (0-127) [optional] // only set if platform requires RX pin number
7  txPin            (0-127) [optional] // only set if platform requires TX pin number
8  dirPin           (0-127) [optional] // only set if half duplex data flow pin is required
6|8|9 END_SYSEX     (0xF7)
````

and then just make the changes in [Configuratble Firmata](https://github.com/firmata/ConfigurableFirmata/blob/master/src/SerialFirmata.cpp): 

- [SERIAL_CONFIG](https://github.com/firmata/ConfigurableFirmata/blob/master/src/SerialFirmata.cpp#L54)
- [SERIAL_WRITE](https://github.com/firmata/ConfigurableFirmata/blob/master/src/SerialFirmata.cpp#L126)
- [SERIAL_READ](https://github.com/firmata/ConfigurableFirmata/blob/master/src/SerialFirmata.cpp#L139)

If that sounds reasonable, I'll give it a shot.

soundanalogous, 2016-06-06:
Seems this could work. The implementation will be tricky however, because you'll need to track dirPin with the portId in order to support `READ_CONTINUOUSLY` mode.

dtex, 2016-06-06:
I was thinking I can just track it in an array, the same way that serialBytesToRead is tracked by port.

soundanalogous, 2016-06-06:
yeah it may be time to add an array of structs

dtex, 2016-06-06:
Structs like this?

````
struct port
{
    int serialBytesToRead,
    unsigned long lastReceive,
    unsigned char maxCharDelay,
    int lastAvailableBytes,
    byte rxPin,
    byte txPin,
    byte dirPin
};
````

soundanalogous, 2016-06-06:
yeah but without the rx and tx pins since those are simply passed to the SW Serial constructor