firmata/protocol #63: How to introduce the protocol to China

nisen, 2016-06-14:
I want introduce the protocol to China maker community. the firmata must have a China name. Chinese General has three:  name, trash name, nickname. 

The arduino to now there is no Chinese name, what many people do not knowThe android's Chinese name is the name of 安卓, was also called early 安智, 安致, etc., with the Chinese name before letting more people know.

## I now translate  firmata to Chinese Name

*  name: 冷杉之眼
*  trash name: 杉树疤
*  杉之芝: 冷杉之上长的灵芝, 

中国名一般有三个,大名,贱名,小名.  还有字,号,别号等 --这些太复杂了. 
比如git的大名是改塔, 贱名是鬼塔, 小名是 吉塔 .

soundanalogous, 2016-06-14:
One way we could do this is to add a folder that includes all of the protocol files translated into Chinese. The folder name would be in Chinese.

nisen, 2016-06-19:
## What We Doing

We fork this [firmata/protocol](https://github.com/firmata/protocol) at <https://github.com/osokay/firmata-protocol-zh>. We translate a little, the 0.0.1. 

I suggest the fold name is zh_cn. Other languages can use its directory.
我建议目录名是zh_cn, 其他语言能使用自己的目录.

I'm not one pair of a translation. I hope the Chinese developers can quickly read Chinese documents, fast start.
我不是一句对一句的翻译. 我希望中国的开发者能快速的阅读中文文档, 快速的开始.