firmata/protocol #64: Step motor 1µs delay in step motor protocol is not accurate

cwssoftwarejoshwright, 2016-07-14:
I have a NEMA 17 step motor.  It runs really fast when I use a 100µs step delay using a standard arduino IDE step motor sketch (aka turn pins on and off with a microseconddelay()).  However when I use configurable firmata (with Digital Inputs, Digital Outputs, Analog Inputs, Analog Outputs, Stepper, Serial, Encoder, and Reporting) with the default 1µs delay, the step motor only moves relatively slowly.

I believe that this is probably due to the time required for Firmata to go through one loop.  Am I correct in thinking that the "real step delay" = "default 1µs delay" + "time required to complete one loop"?  And if so, is there any way to decrease this delay to make it closer to 1µs.

soundanalogous, 2016-07-14:
The 1µs is only used for the stepper drive and only between 2 concurrent steps: https://github.com/firmata/ConfigurableFirmata/blob/master/src/utility/FirmataStepper.cpp#L380-L387. This runs once per update. The udpate time depends on how much work happens in the main loop. A better implementation would be interrupt-driven, but that is challenging in Firmata because other features may also use interrupts. It is something I'm considering for a future release however. I'm also considering switching to use the AccelStepper library since it's better supported than the crappy stepper driver I wrote.

cwssoftwarejoshwright, 2016-07-14:
Awesome, thanks so much!

cwssoftwarejoshwright, 2016-07-14:
So I found a hardware based workaround.  I am using a Big Easy Driver in a two-wire configuration.  The default micro step resolution is 1/16th of a step.  However, you can change the resolution by changing the MS1, MS2, and MS3 pin states using HIGH and LOW.  To find the appropriate combination of states for the desired step size the microstep resolution truth table can be found at:

https://learn.sparkfun.com/tutorials/big-easy-driver-hookup-guide

I have changed my microstep resolution to full step and can now rotate the step motor at the maximum speeds that it can handle.  

My only question now is the following: How is the max speed parameter calculated? For example I am using a JavaScript implementation of Firmata and the following command to move the step motor:

board.stepperStep(deviceNumber, direction, numberOfSteps, stepMotorSpeed, acceleration, deceleration, function() {
            console.log("Step Motor Movement Complete");
});

Now I want the stepMotorSpeed to be maxed out.  Right now I have it set to an arbitrarily high number like 50000, so this does the trick.  However, I would like to understand how the max motor speed is determined so I have a better understanding of what range of values is possible (aka 1 - X?). In other words, at what number does increasing the stepMotorSpeed no longer have an impact?

soundanalogous, 2016-07-14:
I don't have the answers. I barely understand the stepper code I wrote. I got most of it from a white paper. This is why I want to change the implementation to use AccelStepper instead of the FirmataStepper class I wrote.

soundanalogous, 2016-07-14:
There is also an issue open to improve the Firmata Stepper API, so if you have ideas for improvement, please chime in here: https://github.com/firmata/protocol/pull/24. I'm also hoping someone will volunteer to take on this work.

GeoMetJosh, 2016-07-14:
Sounds good, Ill see what I can come up with.  Thanks!

soundanalogous, 2016-10-22:
See https://github.com/firmata/protocol/issues/42. I think the updated protocol and switch to AccelStepper will resolved this issue.

closed 2016-10-22 by soundanalogous