firmata/protocol #65: update I2C feature to allow multiple I2C ports

soundanalogous, 2016-07-16:
It should also use a single command ID. A new command ID can be allocated so the old I2C version can be safely deprecated.

soundanalogous, 2017-01-22:
Also provide a way to frame data so that it can be sent over a series of commands in the event that the buffer exceeds the 64-byte limit.

PizzaProgram, 2020-03-24:
**Hi,**
 Is this feature implemented / finished / stable?
At [this doc](https://github.com/firmata/protocol/blob/master/i2c.md) it is confusing that the 1th line says: "Currently **only** supports **one** I2C port per board"
...
 but a few lines under it:
_"A firmware implementation should support read continuous mode **for several I2C devices simultaneously**"_

I'd like to control multiple USB+Ethernet Ardruino NANOs + multiple extended boards attached to their I2C via OpenHAB.

**Thanks :)**
PS: _Seems to be a GREAT WORK ! Congratulation._

soundanalogous, 2020-03-25:
Only 1 I2C port is supported. You can connect multiple I2C devices to a single port as long as you don't have address conflicts.

PizzaProgram, 2020-03-25:
Thank you VERY much for your answer!
One I2C BUS can handle max **6x (5V)** or 10x (3.3V) extension boards, so 1 I2C bus is enough for me :)

Can you please **extend the documentation** about:
- What is the default bus number? 0? 1?
- Is it auto-discovering the attached I2C boards?
- What speed is it communicating by default? (100K of I2C >> forwarding with ? K to USB/ETH)

soundanalogous, 2020-03-26:
> What is the default bus number? 0? 1?

I assume 0. Seems architecture / framework specific. For an Arduino implementation, it's whatever the default I2C bus is.

> Is it auto-discovering the attached I2C boards?

No, this seems like something better handled at the application level rather than the protocol level.

> What speed is it communicating by default? (100K of I2C >> forwarding with ? K to USB/ETH)

Not enforced by the protocol so the speed depends on the implementation. So for the Arduino implementation it's whatever Arduino defaults to for I2C.