firmata/protocol #76: Enable reporting of analog input pins > 15 and/or values > 14 bits

soundanalogous, 2016-12-10:
Although not strictly limited by the Firmata protocol, currently Firmata implementations allow writing values to analog pins > 15 or analog (pwm) values > 14 bits using the [EXTEND_ANALOG](https://github.com/firmata/protocol/blob/master/protocol.md#extended-analog) message. Reading analog input pins > 15 and/or analog input values > 14 bits should also be enabled.

There are a couple of options:

1. Use EXTENDED_ANALOG to report both analog input and analog output values:

```
0  START_SYSEX              (0xF0)
1  extended analog message  (0x6F)
2  pin                      (0-127)
3  bits 0-6                 (least significant byte)
4  bits 7-13                (most significant byte)
... additional bytes may be sent if more bits are needed
N  END_SYSEX                (0xF7)
```

2. Create a separate message for reporting analog input values and repurpose EXTENDED_ANALOG to only report analog output (pwm) values (as it has traditionally been used):

```
0  START_SYSEX              (0xF0)
1  extended analog write    (0x6F)
2  pin                      (0-127)
3  bits 0-6                 (least significant byte)
4  bits 7-13                (most significant byte)
... additional bytes may be sent if more bits are needed
N  END_SYSEX                (0xF7)

0  START_SYSEX              (0xF0)
1  extended analog read     (0x66)
2  pin                      (0-127)
3  bits 0-6                 (least significant byte)
4  bits 7-13                (most significant byte)
... additional bytes may be sent if more bits are needed
N  END_SYSEX                (0xF7)
```

In either case, this would be accompanied by the ability to enable analog input reporting on pins > 15: https://github.com/firmata/protocol/issues/68#issuecomment-258748963

soundanalogous, 2016-12-14:
TBD if analog pin count should start from 0 or should start from actual pin number (for example analog 0 is actually pin number 14 on most Arduino boards).

zfields, 2017-02-27:
You already know my vote! ;-)

Just to reiterate, it feels like you are doing a lot of extra work to use a nibble to adhere to the `ANALOG_MESSAGE` format. It is more bitwise efficient (so it deserves to remain), but it is hard-capped at `0xF` and requires mapping to work correctly (as you mentioned [Arduino natively supports](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L56-L72) the actual pin number).

If you look further down the Boards.h file, you can see how it doesn't scale well and starts to break down when the analog pins are in non-contiguous blocks. As boards become more diverse, it seems like `EXTENDED_ANALOG` is in a better position to adapt going forward.

Finally, this mapping is only available to clients through the protocol. The mapping requires a priming query, memory allocation and run-time mapping (which can be error prone and is far from efficient - code size or memory footprint).

soundanalogous, 2017-02-27:
Yeah for Firmata 3.0 it probably makes the most sense to simply deprecate `ANALOG_MESSAGE` and fully switch to `EXTENDED_ANALOG`. TBD whether or not to use the `EXTENDED_ANALOG` message for both reading and writing (PWM and possibly also shared with DAC).