firmata/protocol #77: Max speed/accel/decel value for stepper

troywweber7, 2016-12-21:
Hoping someone who knows the protocols better can answer this because I could not find hte answer documented here: https://github.com/firmata/protocol/blob/master/stepper-1.0.md.

The max number of steps is recorded in the protocol (3 bytes send, only 21 bits used). Since there are only two bytes sent for speed/accel/decel, does this mean the max speed/accel/decel values are 16-bit or less?

soundanalogous, 2016-12-21:
Yes, max is 14 bits for acceleration and deceleration values. This cannot be changed in a non breaking way. We need to make sure Stepper 2.0 supports an appropriate resolution here.

troywweber7, 2016-12-21:
@soundanalogous and at whomever asked for my input on the new protocol (although that comment appears to have been removed): I have been using 1.0 protocol for several months now in an Auto-Titration machine used by a scientific soils analysis firm. They've prototyped the machine, I've been in charge of automating it. And my experience with controlling steppers, while in infancy compared to some mechatronics engineers, is still relatively intimate.

If you truly want my input, then I'm going to take a hard look at stepper 2.0 protocol and give my feedback on this thread (unless there is another thread I ought to consider, in which case, please link it here).

soundanalogous, 2016-12-21:
Please add to the original Stepper 2.0 thread to keep everything in context: https://github.com/firmata/protocol/issues/42.

troywweber7, 2016-12-24:
> Stepper Motor 2.0
> ===
> 
> Provides support for full 2 wire, full 3 wire, full 4 wire, half 3 wire, and half 4 wire stepper motor drivers (H-bridge, darlington array, etc) as well as step + direction drivers such as the [EasyDriver](http://www.schmalzhaus.com/EasyDriver/).

**EasyDriver is great, but I'd like to recommend that it also support the [BigEasyDriver](https://www.sparkfun.com/products/12859). I'd like to note that it only adds 1 more micro-step pin to the configuration, which allows 1/16 micro-stepping mode. The likelihood of ever needing smaller resolution than this is minimal, however.**

> Current implementation supports 10 stepper motors at the same time (#[0-9]).
> 
> Includes optional support for acceleration and deceleration of the motor.
> 
> Also includes multiStepper support which allows groups of steppers to be simultaneously passed different ```to``` values and the duration of their movements will match. Up to five multiStepper groups can be created. The total number of steppers is still limited to 10.
> 
> Example files:
>  * Version 2.0 of the stepper protocol has not yet been implemented.
> 
> Protocol
> ---
> 
> **Stepper configuration**
> ```
> 0  START_SYSEX                                (0xF0)
> 1  Stepper Command                            (0x62)
> 2  config command                             (0x00 = config)
> 3  device number                              (0-9) (Supports up to 10 motors)
> 
> 4  interface                                  (upper 3 bits = wire count:
>                                                 000XXXX = driver
>                                                 010XXXX = two wire
>                                                 011XXXX = three wire
>                                                 100XXXX = four wire)
> 
>                                               (4th - 6th bits = step type
>                                                 XXX000X = whole step
>                                                 XXX001X = half step
>                                                 XXX010X = micro step)

**This seems odd to me. Micro-step seems to cover 1/4, 1/8, and 1/16 step. Is this intentional? Why not explicitly provide for each? If this is intentional, I would just use bits 4th-5th.**

>                                               (lower 1 bit = has enable pin:
>                                                 XXXXXX0 = no enable pin
>                                                 XXXXXX1 = has enable pin)

**Please note there is also a sleep pin. [I think the enable is more useful](https://staceyw1.wordpress.com/2011/01/08/test/), but if it were me, I'd include the sleep pin functionality just in case. If sleep were included (the the 6th bit were freed up), you could make the 6th bit be "has enable pin" and the 7th bit be "has sleep pin".**

> 5  minPulseWidth                              (0-127 measured in us)
> 6  num steps-per-revolution LSB
> 7  num steps-per-revolution MSB
> 8  maxSpeed LSB
> 9 maxSpeed MSB
> 10 motorPin1 or directionPin number           (0-127)
> 11 motorPin2 or stepPin number                (0-127)
> 12 [when interface >= 0x0110000] motorPin3    (0-127)
> 13 [when interface >= 0x1000000] motorPin4    (0-127)
> 14 [when interface && 0x0000001] enablePin    (0-127)
> 15 [optional] pins to invert                  (lower 5 bits = pins:
>                                                 XXXXXX1 = invert motorPin1
>                                                 XXXXX1X = invert motorPin2
>                                                 XXXX1XX = invert motorPin3
>                                                 XXX1XXX = invert motorPin4
>                                                 XX1XXXX = invert enablePin)
> 16 END_SYSEX                                  (0xF7)
> ```
> 
> **Stepper zero**
> ```
> 0  START_SYSEX                             (0xF0)
> 1  Stepper Command                         (0x62)
> 2  stop command                            (0x01)
> 3  device number                           (0-9)
[... 126 lines cut ...]
> ```
> 
> **MultiStepper stop**
> ```
> 0  START_SYSEX                             (0xF0)
> 1  Stepper Command                         (0x62)
> 2  multi stop command                      (0x23)
> 3  group number                            (0-127)
> 4  END_SYSEX                               (0xF7)
> ```
> 
> **MultiStepper report positions (sent when a move completes or stop is called)**
> ```
> 0  START_SYSEX                             (0xF0)
> 1  Stepper Command                         (0x62)
> 2  multi stop reply command                (0x24)
> 3  group  number                           (0-127)
> 4  member number                           (0-9)
> 5  device 0 position byte 1 LSB
> 6  device 0 position byte 2 MSB
> 
> *Optionally repeat 4 through 6 for each device in group*
> 
> 33 END_SYSEX                               (0xF7)
> ```

closed 2016-12-24 by troywweber7

soundanalogous, 2016-12-24:
>
> *Please note there is also a sleep pin*

This would be best handled outside of the Firmata stepper interface since
using the regular Firmata digital output channel. A Firmata client library
could build this into the stepper API, but it would just use digital
messaging under the hood. The only reason enablePin is included here is
because it's a convenience provided by the AccelStepper library.

Regarding micro-stepping. AccelStepper does not currently support
micro-stepping. If we want that functionality we'll need to contribute that
to AccelStepper. For something like the EasyDriver or BigEasyDriver, the
AccelStepper whole or half step doesn't even apply to those driver boards,
they only apply to 3 and 4 wire stepper configurations. For 2 wire drivers,
the selection of micro stepping would be achieved using regular Firmata
digital messages and would could be wrapped in the client stepper API for
convenience.

The last observation also highlights an issue with the current Stepper 2.0
proposal where the whole/half/micro stepper can't be crossed with the
DRIVER interface type. It would be a no-op.