firmata/protocol #78: Define a new Servo 2.0 interface that is decoupled from AnalogWrite

soundanalogous, 2016-12-24:
The current Servo interface used the Firmata Analog message to set it's value. This creates tight coupling between ServoFirmata and AnalogWrite and AnalogOutputFirmata.

We need a new Servo interface that uses Sysex. Does anyone have an opinion on changes to the existing firmata servo interface?

dtex, 2017-01-02:
I would like to see multiServo that allows me to pass a position for each member in a group of servos in a single message (see multiStepper).

rwaldron, 2017-01-09:
Ability to write in degrees or µsec (the latter is nice for high resolution writing)

dtex, 2017-01-09:
Also worth noting that not all servos are 180 degrees so you'd have to be able to pass in the physical range of the servo as part of a configuration command in firmata. If that happens, I want to be able to specify radians as a configuration option because radians are better than degrees for IK work. Having to convert back and forth slows things down.

zoooofu, 2017-01-12:
I would suggest to add an optional feature ID for servo usage, SERVO_OPERATION for example. Like what I2C_REQUEST having done, the existing servo interface servoWrite(pin, degree) would be one mode of it. However, servoWrite() function can only make servo move as we requesting, we also need to know the state of servo in some cases. So, I think it would be nice to add servoRead() function, which would be one mode of SERVO_OPERATION as well.