firmata/protocol #86: [v3.0.0] PIN_STATE_QUERY should also report current resolution

zfields, 2017-03-22:
```
0  START_SYSEX              (0xF0)
1  pin state response       (0x6E)
2  pin                      (0-127)
3  pin mode (the currently configured mode)
?  pin resolution (the resolution to support the mode) <--------------------- PLEASE ADD THIS
4  pin state, bits 0-6
5  (optional) pin state, bits 7-13
6  (optional) pin state, bits 14-20
... additional optional bytes, as many as needed
N  END_SYSEX                (0xF7)
```

soundanalogous, 2017-03-22:
you can already get the resolution from the capability query

soundanalogous, 2017-03-22:
also, I could add it in a backwards compatible way by adding it after byte 6

zfields, 2017-03-22:
I thought it would tell the maximum resolution the pin is capable of in the capability query. I am looking for the current resolution; similar to pin mode.

soundanalogous, 2017-03-22:
Capability query is the current default resolution for the pin. There is currently no API in the protocol that enables changing the pin resolution (although I have had a proposal open for a number of years). This does get more complex as more boards now have DACs. PWM resolution may be able to change on a per pin basis (for non-Arduino boards). Analog input resolution can currently only be changed globally for Arduino, but perhaps for other platforms it can be changed on individual pins (although Firmata currently only supports Arduino and Particle).

zfields, 2017-03-22:
Fair enough. I just thought I would float the idea. I don't really have a use for it now, but as I complete the wiring API, I can see that it will become necessary.

soundanalogous, 2017-03-22:
It's a worthwhile suggestion. I'm not sure yet whether it fits better into the `pin state query`or if it make more sense as part of the [proposed analog config message](https://github.com/firmata/protocol/pull/8/files) (which needs to be updated in general), in which case I'd add getters for the analog read and write resolutions (global per board rather than pin-based).