firmata/protocol #87: Changes in stepper2

dtex, 2017-03-23:
* Remove speed params in step
* Remove speed params in to
* Change set maxSpeed to speed (Stepper2 will figure out wether to use AccelStepper's setSpeed or setMaxSpeed command automatically, based on the current acceleration value)

I believe speed will rarely need to be passed in a step or to call and probably should be done in a separate command. I had put it in to match the existing stepper_step command, but given that this is a complete rewrite I think it's okay to drop it.

soundanalogous, 2017-03-25:
This sounds good to me as long as you're confident you'll always be able to infer the correct maxSpeed for any compatible stepper driver. I've also moved stepper-2.0.md to /proposals/ since there is still some churn here.

soundanalogous, 2017-06-03:
@dtex are you still planning to make these changes?

dtex, 2017-06-03:
Yes, I have made those changes. I haven't made a PR because there might be a few more things that pop up during the implementation (the multiStepper.stop issue for example).

dtex, 2017-07-03:
PR in #90

closed 2017-09-16 by soundanalogous