firmata/protocol #89: common optional SYSEX commands

monteslu, 2017-06-01:
Thinking it would be cool if we kept some documentation around of SYSEX commands people are using.  Things such as the codes for neopixel interaction in node-pixel or the accelerometers in playground-io.

No set in stone registry, just documentation on what's being used to help avoid collisions or give guidance on implementation for new hardware.

soundanalogous, 2017-06-01:
there is a place to do that here: https://github.com/firmata/protocol/blob/master/feature-registry.md

If it's not in the registry, then it shouldn't be used in any public library. It's the fault of the library author in that case for not getting it into the registry.

soundanalogous, 2017-06-03:
These types of hardware-specific features should also use the Extended ID (`00H nnH nnH`). For example we can add IDs for the accelerometers in playground-io, but they have to be the 3 byte variety.