firmata/protocol #94: Firmata Host implementation for Linux

dimitry-ishenko, 2017-09-19:
It would be nice to have a Firmata host side implementation for Linux. There are plenty of SBCs running Linux that could benefit from it. Few immediate ones that come to mind:

[Raspberry Pi](https://www.raspberrypi.org/)
[CHIP](https://getchip.com/)
[VoCore2](http://vocore.io/)

By leveraging [GPIO](https://www.kernel.org/doc/Documentation/gpio/) and [Device Tree](https://www.kernel.org/doc/Documentation/devicetree/) facilities the host can probably even auto configure itself.

dimitry-ishenko, 2017-09-19:
My immediate interest is being able to control I/O remotely through WiFi using the VoCore2 chip.

monteslu, 2017-09-20:
take a look at remote-io:  https://github.com/monteslu/remote-io
It can act as a firmata server.  Think of it as the opposite of firmata.js

You take any johnny-five IO class and wrap it with remote-io, and provide it some type of transport such as a tcp or websocket.  Most of the functionality is there for GPIOs. Digitial inputs might need a little work, but outputs, pwm, analog in, and i2c should all work.

slightly outdated, but I did a writeup of it here:  https://www.iceddev.com/blog/remotebots-cutting-the-cord-with-remote-io/