firmata/protocol #102: Firmata sending the state fof all inputs when just one input is triggered.

jupiter8888, 2018-04-26:
i'm having this issue in Node-Red both with the arduino node and the contrib-gpio node wich use firmata, and a raspberry pi. Tried also with Node-red on a pc and on an orange pi, same result.

Here's the thing:

 I have a digital only board filled with RC circuits on every inputs. (12 of them on an Uno) and if i switch, say input 4, the GPIO node 4 gives "0" (true),  that's normal,  BUT,  EVERY other  GPIO nodes give 1 (false, wich is their actual state)  so all the pins are reporting their state on state change of one pin.

the node says "It only outputs on a change of value", is this a bug or i'm doing something wrong here?

Thanks.

dtex, 2018-04-26:
firmata sends digital reads in groups of 8 to the host (firmata.js in this case). This happens when any one pin in the group changes. firmata.js loops through the 8 pins (aka the port) and emits a ```digital-read-N``` where N is the pin number for each of those 8 pins whether they have changed or not.

It sounds like firmata is behaving as expected and you need checks in node-red or your code to make sure action is only taken when the value on the relevant pin changes. This means tracking pin state. Have you asked this question over in the node-red community?

monteslu, 2018-04-26:
might want to try running a test with just firmata.js.  Both node-red-contrib-gpi and node-red-arduino use firmata.js under the hood, so not sure why multiple events would be triggered unless firmata.js isn't tracking state changes correctly

dtex, 2018-04-26:
@montesul I don't think that's something firmata.js tries to do. In J5 we are tracking state on each instance (button, etc) and deciding whether or not to fire there.

Here are the relevant lines:

https://github.com/firmata/firmata.js/blob/master/lib/firmata.js#L124-L149