firmata/protocol #108: Introduction to community and query regarding any efforts/interest in CAN as transport

rjrodd, 2018-07-06:
Hello!

I recently became aware of Firmata and have been enjoying using it for Raspberry Pi 3 <-> Arduino* communication over the past couple of months.  [* Ether10s and EtherMegas, firmata.js on the Pi's ]

In an effort to reduce power consumption I would like to replace the Ethernet transport layer (and related router/shield infrastructure) with CAN infrastructure.  I'm happy to develop that software component for the Arduino and NodeJS stacks; just curious if anyone else has already explored/implemented it or has tips to offer.

Relatedly, is there already existing or is there any interest in using a Firmata Slack group? .. or perhaps just a channel in the NodeJS group ( joined via http://www.nodeslackers.com/ ).

Thanks!

Feel free to close this, or I shall, once most of the active Firmata developers/enthusiasts have a chance to chime in on CAN thoughts.

soundanalogous, 2018-07-07:
Hi, AFAIK there have been no proposals for a CAN transport layer. If this is something you are interested in contributing, take a look at the Ethernet , WiFi and BLE transports and see if you can design a CAN transport that works in a similar way. I'm not very familiar with the details of the CAN protocol myself so I'm not sure how compatible it is with Firmata.

soundanalogous, 2018-07-07:
Regarding a Firmata Slack group. There is a Firmata Slack group with only 2 users (myself and zfields). It's pretty quiet though as I'm not really a Slack user (trying to keep distractions to a minimum) - I can't even figure out my Slack login ATM. Firmata development in general is pretty quiet and slow moving.

rjrodd, 2018-07-31:
Okay, thanks!  If you ever find the name of the Firmata Slack workspace / group / team (whatever Slack calls it) I would be eager to join in the discussions.  Regarding CAN transport .. I have something in place at the moment working on virtualized-CAN (Linux's `socketcan` library) and will be doing hardware testing before too long.

closed 2018-07-31 by rjrodd

monteslu, 2018-07-31:
@rjrodd hasn't been active in a while, but there is https://gitter.im/firmata/protocol