firmata/protocol #111: [question] Sysex ids

martin-eden, 2019-05-22:
Hello, I'm implementing Firmata parser and have some questions regarding extending sysex format:
1. Which byte order is assumed?
  In (00 12 34), (12) is low 7 bits of ID or high?
2. What if extended ID matches one-byte ID?
  Assuming low-byte-first, consider (69) vs (00 69 00). 
  Is it same ID? Should such duplicates be forbidden or allowed?

BTW in [feature-registry.md](https://github.com/firmata/protocol/blob/master/feature-registry.md) sysex extended ID bytes 2..n described as 

```EXTENDED_ID (00H 00H - 7FH 7FH) + PAYLOAD```

I think it should be

```EXTENDED_ID (00H - 7FH 00H - 7FH) + PAYLOAD```

Best wishes.

soundanalogous, 2019-05-23:
The left byte is the low byte, so in the case of extended ID a low byte of 00H indicates you should expect the 2 following bytes to be the extended ID. However there are no features using the extended ID so parsing it provides no value at this time.

See this section of prototcol.md for more info: https://github.com/firmata/protocol/blob/master/protocol.md#sysex-message-format.

soundanalogous, 2019-05-23:
You can also inspect the c++ parser implementation here: https://github.com/firmata/arduino/blob/master/FirmataParser.cpp

martin-eden, 2019-05-23:
I'm implementing parser based exactly on that `protocol.md`. And I consult mentioned C++ implementation for shady moments. Problem is that C++ implementation don't support extended IDs.

Original questions remains: byte order in extended ID (LSB or MSB), equality policy ((F0 69 F7) ==? (F0 00 69 00 F7) ).

soundanalogous, 2019-05-24:
(FO 69 F7) != (F0 00 69 00 F7)

The extended ID must be 3 bytes. The first byte is the command byte 00, this signals that it's an extended ID. The next 2 bytes define the actual ID value, so the value is 0x6900H. Even if you set the Extended ID to 0x0069H, it still would not be the same as a SysEX command value of 0x69H. I guess the term "Extended ID" is misleading here since it's not really an extension of the 7-bit command value range, it's really a separate 14-bit range of IDs.

soundanalogous, 2019-05-24:
I adapted this from the midi extended manufacturer ID scheme. You can see here that the extended manufacturer ID != to the same ID in the normal 127 bit range: https://www.midi.org/specifications-old/item/manufacturer-id-numbers.

martin-eden, 2019-05-24:
Thank you! It clear now.

closed 2019-05-24 by martin-eden