firmata/protocol #112: Incorrect value for "move complete command" in AccelStepper?

pavhofman, 2019-11-25:
The "move complete command" in AccelStepper lists value 0x0b in 
protocol/accelStepperFirmata.md  . However, the implementations use 0x0a:

https://github.com/firmata/firmata.js/blob/488964da379b137da5ab3e7887161cbf31c60793/packages/firmata-io/lib/firmata.js#L400

https://github.com/firmata/ConfigurableFirmata/blob/master/src/AccelStepperFirmata.h#L39

Since the arduino firmware uses 0x0a, please can you fix the documentation? 

Thanks a lot.

dtex, 2019-11-25:
You are absolutely correct. I must have removed a command at some point during the implementation and forgot to update that value. I'll get a PR in ASAP.

pavhofman, 2019-11-25:
Thanks a lot for instant response :-)

dtex, 2019-11-25:
Thank you for the catch! 

It's all updated now.