firmata/protocol #115: Extend documentation

PizzaProgram, 2020-03-25:
Dear Developers!
 It is very helpful to read details for those, who are just beginning to understand what is this all about. It can save dozens of hours for everyone.
That's why I'm asking:
- Please **extend** the documentations even more!
_(Every extra word / example / URL-link to already answered issues: helps a lot.)_

Infos about:
- --
0. Is this a "master" or "slave" program or both / choosable / set-able?
    - (it's not really clear / defined what part you call "client"?)

1. What timing does it use to communicate to each other?
    - synced / async?
    - what rate?
    - continuously / only triggered / timed / choosable ? 

2. How to handle / distinguish multiple boards attached to the same master?

3. How to scan / read devices on a I2C bus?
    - (Or is it forwarding the I2C Bus directly to the master?)

4. What is the difference between StandardFirmata and StandardFirmataPlus?

5. Which libraries are frequently maintained / tested / stable?
    - most recommended.

Thank you very much for this awesome work in the name of all of us!

luvarpt, 2020-07-06:
I would also like some info about goal of protocol. That it would like to be more efficient than ASCII based protocols, it counts on relatively low speed communication and also it does limit (not hard limit) messages length to have a good latency of commands. I.e. do not block communication channel for too long with single command. At least link to wiki can be helpful http://firmata.org/wiki/Design_Issues

tochinet, 2022-02-14:
Yes, most open source projects are not the strongest in documentation. Nobody likes to spend time with it. Worst, many people just "fork" (make their own copy), change it for their immediate need, and ... forget altogether that their experience could be useful for others.

@luvarpt , on the "goal of the protocol", the basic idea is that the Arduino has lots of input outputs, but is limited in power. So  instead of cramping the logic of the program in a 32kB ROM / 2kB RAM, you put there a generic program, and you make a program on a PC/PI/whatever in whatever language that says "put pin8 high" and it does. So the Arduino with Firmata is a dumb "USB/Serial to Physical" bridge. 

I looked at Firmata back in 2011 and abandoned it because what it actually does is move your problem ack to the PC, but the beauty of Arduino is that it repeats the program when you power it up. Not so with Firmata, since you make it dumb.

@PizzaProgram here are my short answers to your questions : 
0:  there is both one slave (the Arduino "server") and many master/client implementations (on PC, usinc C, javascript, processing, whatever).
1: it normally uses Serial, speed can be adapted to your wish (57600bps on StandardFirmata.ino)
2. Serial is point to point, you DON"t attach multiple boards with Firmata. Except using multiple Serial interfaces, but that's the FirmataPlus
3. Yes, it bridges the I2C to the master, so the master can only send:receive I2C commands. Maybe there is a I2Cscan capability, I honestly don't know
4. StandardFirmata nad StandardFirmataPlus are _examples_ of usage of the Firmata library. Basically the plus (it is written in the example) means that the Arduino itself also has serial interfaces to extend the capabilities. So you have one PC with one Arduino via Serial, and additional Serials to "second-level" boards. Not sure this can be very efficient as compared to I2C or SPI to be honest...
5. The server on Arduino gets revised and augmented regularly (yearly ?) on Github, look at the dates and speed to respond to see if the community is alive. If you get in 2022 a response to a simple question of March 2020, it's not a very good sign, but there is worse... (No flames please, you had 2 years to be faster than me, I am not interested in your critics)