firmata/protocol #140: High Resolution PWM control

zxia545, 2022-11-19:
I have a problem with the current protocol to control Arduino to generate high-resolution PWM. Since some Arduino can config some timers from 8-bit to 16-bit such that PWM can be controlled more accurately. However, due to the current protocol design, it can only transfer 14-bit of data for PWM control. 

In my project, I have to change Arduino Firmata Express and Python Pymata4 to achieve a maximum transfer of 21-bit to achieve this case of use. 

Just wondering if anyone else needs this feature.

pgrawehr, 2022-12-23:
I'm not sure I understand your requirement. Do you need a higher resolution of the duty cycle (e.g 16 bits) or a way to set the frequency (for which there's currently no option in the protocol)?

zxia545, 2022-12-23:
> I'm not sure I understand your requirement. Do you need a higher resolution of the duty cycle (e.g 16 bits) or a way to set the frequency (for which there's currently no option in the protocol)?

Yap, I need a high-resolution duty cycle and the current protocol does not directly support it. (But it is not a problem, can be easily modified to achieve that )

pgrawehr, 2022-12-23:
> Yap, I need a high-resolution duty cycle and the current protocol does not directly support it. 

That's not exactly correct. To use more than 14 bits for the PWM resolution, you can use the EXTENDED_ANALOG message (described [here](https://github.com/firmata/protocol/blob/master/protocol.md#extended-analog)) instead of the fixed-length ANALOG_MESSAGE. To take advantage of this, you need a board that supports the higher resolution, of course.

zxia545, 2022-12-23:
> 

> That's not exactly correct. To use more than 14 bits for the PWM resolution, you can use the EXTENDED_ANALOG message (described [...]

I used an Arduino MEGA2560 and I can change some registers to let it support higher resolution. After that, it becomes a little bit different since I cannot directly call Arduino **analogWrite(**) method.

pgrawehr, 2022-12-23:
`analogWrite()` takes an int (16 bit) as the second argument, so this should actually work. Have you created a test sketch that uses `analogWriteResolution` and `analogWrite` to test how this works with 16 bits?